fm4dd/gatemate-riscv #6: step19 make vtest on verilator version: erilator 5.020 2024-01-01 rev (Debian 5.020-1) not working

Elektronikus, 2024-10-17:
iverilog simulation with make test is working fine,
verilator compiles ok, but than reports errors:
Incorrect label initialization
Expected: 12    Got: 0
Label:         12
Incorrect label initialization
Expected: 16    Got: 0
Label:         16
Incorrect label initialization
Expected: 76    Got: 0
Label:         76
Incorrect label initialization
Expected: 84    Got: 0
Label:         84
Incorrect label initialization
Expected: 96    Got: 0
Label:         96
Label:        188
Label:        264
Incorrect label initialization
Expected: 272    Got: 0
Label:        272
Label:        284
Incorrect label initialization
Expected: 292    Got: 0
Label:        292
Label:        308
Incorrect label initialization
Expected: 316    Got: 0
Label:        316
Label:        328
Label:        344

fm4dd, 2024-11-20:
Hi Elektronikus,

I did not get far on it, but this warning started from verilator version 5.017 onwards. I found a reference in step19/SOC.tb:
```
TD_SOC_tb.uut.RAM.Label ; 
    %load/vec4 v0x562f03ec3ef0_0;
    %parti/s 1, 0, 2;
    %cmpi/e 1, 1, 1;
    %jmp/0xz  T_43.6, 6;
    %vpi_call 6 613 "$display", "Missing label initialization" {0 0 0};
    %pushi/vec4 1, 0, 32;
    %store/vec4 v0x562f03ecb420_0, 0, 32;
    %jmp T_43.7;
T_43.6 ;
    %load/vec4 v0x562f03ec3ef0_0;
    %load/vec4 v0x562f03ecbcb0_0;
    %cmp/ne;
    %jmp/0xz  T_43.8, 4; 
    %vpi_call 6 616 "$display", "Incorrect label initialization" {0 0 0};
    %vpi_call 6 617 "$display", "Expected: %0d    Got: %0d", v0x562f03ecbcb0_0, v0x562f03ec3ef0_0 {0 0 0};
    %pushi/vec4 1, 0, 32; 
    %store/vec4 v0x562f03ecb420_0, 0, 32;
T_43.8 ;
T_43.7 ;
    %vpi_call 6 620 "$display", "Label:", v0x562f03ecbcb0_0 {0 0 0}; 
    %end;
S_0x562f03ec3ff0 .scope function.vec4.s32, "LabelRef" "LabelRef" 6 625, 6 625 0, S_0x562f03eb51a0;
 .timescale 0 0;
v0x562f03ec41d0_0 .var/i "L", 31 0;
; Variable LabelRef is vec4 return value of scope S_0x562f03ec3ff0
```
The output traces into step19/obj_dir/VSOC___024root__DepSet_hfcbc957c__0__Slow.cpp.

Elektronikus, 2024-12-27:
Hi,
I'm quite lost. 
How did the load command look in version 5.016?
 %load/vec4 v0x562f03ecbcb0_0; ???