fm4dd/gatemate-riscv #7: Resource Utilization

PythonLinks, 2025-01-03:
Could you please be so kind as to state the resource utilization for 
these RISC-V cores on gatemate?

fm4dd, 2025-01-04:
Hi Christopher,

In the repo tutorial steps, the readme's should show the Yosys synthesis hardware utilization report. As an example, below is the utilization report of [step20](https://github.com/fm4dd/gatemate-riscv/tree/main/step20):

```
   Number of wires:                590
   Number of wire bits:           3187
   Number of public wires:          67
   Number of public wire bits:    1041
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               1214
     CC_ADDF                       170
     CC_BRAM_20K                     5
     CC_BUFG                         1
     CC_DFF                        106
     CC_IBUF                         3
     CC_LUT1                        37
     CC_LUT2                        50
     CC_LUT3                       388
     CC_LUT4                       444
     CC_OBUF                         9
     CC_PLL                          1
```

Putting this data into relation to the Gatemate CCGM1A1 resource specs, it looks like this:

The Gatemate CCGM1A1 FPGA has a total of 20,480 CPE, configurable as 20,480 8-Input-LUTs or 40,960 4-Input-LUTs/FF/Latches. From the report above, I count the LUT1..4, DFF and ADF elements, which adds up to 1195 of 40,960 LUTs/FF/Latches. It is probably appropriate to add the input/output buffers, which raises the count to 1209 elements (to small to measure a an increase in utilization percentage).

 The logic element utilization is about 2.9%. For the RAM and PLL elements, the tutorial design uses 1 of 4 available PLLs (25%), and 5 of 64 available 20K BRAM blocks (8%).

The authors RV32I RISCV tutorial is a minimal implementation that was implemented on an entry-level IceStick Ice40HX1K with only 7680 programmable logic cells and 128 kbits BRAM. In comparison, the Gatemate CCGM1A1 FPGA has 5 times more logic elements, and about 10 times more BRAM.

For this tutorial, the Gatemate CCGM1A1 FPGA has ample resources to run the basic RV32I RISCV core implemented by Bruno Levi. I am also confident it can run the advanced sequel tutorial "Episode II", which adds a sequential 5-stages pipeline and adds the RV32M extension (not yet tested).

I also haven't tried other RISCV core implementations, but I think the CCGM1A1 can handle a range of common RISCV implementations. The bigger Gatemate CCGM1A2 FPGA (80k LUT/FF) should be even more potent.

Cheers!
Frank

Reference: [Gatemate datasheet](https://colognechip.com/docs/ds1001-gatemate1-datasheet-latest.pdf).

PythonLinks, 2025-01-04:
Thank you.  That was most helpful.  That is about the same number of 4 input LUTs that the FEMTORV uses on the ICE40 devices.  That makes sense.  

IIRC, the FemtoRV stores the registers in memory, and just loads the two source registers.  With the multiplexers in each Cologne Programmable Element (CPE), it might be possible to move the register file into CPEs.  

Just a thought. 

Thanks for porting this repository over.  

I linked to this repository from 
https://github.com/PythonLinks/awesome-gatemate

fm4dd, 2025-01-05:
Hi Christopher,

Thank you for the kind comments, and for sharing the link to your repository. Creating a curated list of all relevant Gatemate resources is great! It is really helpful to keep track of the Gatemate FPGA ecosystem that is building up, especially in the open-source community.  This reference is much needed!

Cheers,
Frank

P.S. You are right, the logic element count is comparable to ICE40 devices. For [step20](https://github.com/fm4dd/gatemate-riscv/tree/main/step20), I checked the Yosys utilization report for an iCEBreaker FPGA ([iCE40 UltraPlus 5K](https://www.latticesemi.com/products/fpgaandcpld/ice40ultraplus)):

```
Info: Device utilisation:
Info: 	         ICESTORM_LC:  1107/ 5280    20%
Info: 	        ICESTORM_RAM:    16/   30    53%
Info: 	               SB_IO:    12/   96    12%
Info: 	               SB_GB:     5/    8    62%
```

The same Yosys report for an ULX3S FPGA board ([Lattice ECP5 LFE5U-85F](https://www.latticesemi.com/products/fpgaandcpld/ecp5)) returns similar counts:

```
=== SOC ===

   Number of wires:                771
   Number of wire bits:           3490
   Number of public wires:         771
   Number of public wire bits:    3490
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               1419
     $scopeinfo                      4
     DP16KD                          5
     LUT4                         1304
     TRELLIS_FF                    106
```

Very roughly, I think the Gatemate FPGA CCGM1A1 resources compare well to Lattice ECP5-5G (LFE5UM5G-45), or AMD's Spartan 6 (XC6SLX45T).

closed 2025-01-05 by fm4dd